Open-Industry-Project/Open-Industry-Project-Godot #26: License: GPL to LGPL with exemption please

echads, 2023-09-21:
GPL doesn't make sense - Unity is closed source (they do have source for the editor on github but it isn't licensed in a GPL compatible way)

LGPL would make sense if want to stick to requiring contributions back.

Here is an example with exemption text:
https://github.com/mcs-sdk/mcs/blob/master/LICENSE.txt

Topic: https://github.com/mcs-sdk/mcs/issues/4

Unity forum topic: https://forum.unity.com/threads/released-morph-character-system-mcs-male-and-female.355675/page-72#post-4584619

TheColonel2688, 2023-09-21:
I agree GPL is cancer.

ryevdokimov, 2023-11-09:
This project is being ported into Godot to avoid licensing issues and to have more flexibility with modifying the engine and editor.

closed 2023-11-09 by ryevdokimov

ryevdokimov, 2023-11-22:
Interesting read. Have you read the response from the Godot lead dev? 

[https://gist.github.com/reduz/cb05fe96079e46785f08a79ec3b0ef21](url)

Long story short I don't doubt that GDScript is not a performant language. This is true of most dynamically typed languages like Python. However, this project will only use C#, but I understand that the engine was originally optimized for GDScript as that article pointed out. A lot of effort in upgrading the 3D and C# elements after the 4.0 update, and if you look at the proposals in their repo, I expect many more optimizations to come. The engine recently got a lot of attention and funding, and I have faith that they will continue to move in the right direction. 

Additionally, performance isn't much a concern at the moment, I've done some preliminary bench testing, and it would take an incredibly complicated simulation to start noticing the effects. Somewhere in the order of thousands of objects interacting on the scene. I am not saying that something like that can't be developed, but since we are still in the earlier stages of the project, I think both the engine and this project have plenty of room to grow. Also depending on how much interest for their project there is, I would be willing to help the Godot community as well. I think it would be great to have a successful open-source game engine that does for its community like Blender did for theirs.

echads, 2023-11-22:
I learned a lot too. I was glad the author had the update at the top.
[Quote from link:](https://gist.github.com/reduz/cb05fe96079e46785f08a79ec3b0ef21#only-a-pathological-use-case-is-shown-the-rest-of-the-api-is-fine)
`Eventually, C# will be moved to the universal extension system and this will allow the unifying of the default and .net editors, it is just not the case yet, but its top in the list of priorities.` _You may not need the performance but good to know it is on the roadmap._ Also this was a good section. `I want to insist that the article written by Sam Pruden is fantastic, but if you are not familiar with how Godot is intended to work under the hood it can be very misleading.`
Great insight into the state of the engine from two sources and what can be done to improve things. Also hopeful they can count you among the contributors too. :-)

ryevdokimov, 2024-02-17:
I've been working on it: https://github.com/godotengine/godot/pulls/ryevdokimov, there remains a lot of work to be done, but it's been good enough to get the job done. I'm optimistic about the direction Godot is going.

echads, 2024-02-21:
That's awesome! What has been your experience like thus far with the godot dev community?

I didn't realize you would have to make adjustments to godot itself. Looks like valuable contributions! I've always wanted to dual develop on both Unity and Godot. There are some RPG frameworks that work on both and the other day I found a physics thing that was a ported GDscript to work with rigid bodies the same way.

I think a lot of devs wish they could justify the switch but a lot of existing code base is in Unity. Switching to open source code is highly desirable for collaboration. So it's tech debt and vendor lock in that are the biggest obstacles IMO.