Open-Industry-Project/Open-Industry-Project-Godot #31: Modbus/TCP connection method

qwert3356, 2024-04-25:
i see the source code inside demo.tscn

the sensor (photo eye) or actuator(conveyer) seems can be connect to EtherNet/IP tag by using libplctag.NET Tag class,
and value will corrsponse to tagName

```
Tag<RealPlcMapper, float> tag = new()
		{
			Name = tagName,
			Gateway = Gateway,
			Path = Path,
			PlcType = PlcType,
			Protocol = (Protocol?)Protocol,
			Timeout = TimeSpan.FromSeconds(5)
		};
```

if the tag Name is conveyer's object name
how to make conveyer connect to certain modbus coil address?

ryevdokimov, 2024-05-01:
The tag name should correspond to the address used in the PLC, not the name of the parts.

qwert3356, 2024-06-25:
sorry for late response,
The PLC we use is mitsubishi FX5U, it can provide classic modbus address like coil 0~40000

could libplctag's modbus TCP support this type of modbus address? 

[image]

ryevdokimov, 2025-06-19:
Communication logic has been redone see: https://github.com/Open-Industry-Project/oip-comms

closed 2025-06-19 by ryevdokimov