Open-Industry-Project/Open-Industry-Project-Godot #77: Fix Jolt scale errors

cooperra, 2024-10-15:
We currently patch Jolt to hide some error messages, but we should fix them for real.

See: https://github.com/Open-Industry-Project/godot-jolt/commit/d6dbe05d736353f81d893926a7dcdffc76bdb0d4

An example of the messages:
[image]

I found some insight about what the error is about here: https://github.com/godot-jolt/godot-jolt/pull/925

In short, inaccuracies with the scale. Probably from accumulated floating point errors in rotation. So if we track down the source of the inaccuracies in our code, we should be able to resolve the errors.

ryevdokimov, 2024-10-15:
Related: https://github.com/Open-Industry-Project/Open-Industry-Project/issues/73

Ideally, we shouldn't be scaling collision/physics bodies at all, but only the mesh, and size of the collision shapes. Without the above proposal though there isn't an easy way to do it.

cooperra, 2024-10-16:
So, I poked at this a little. It's just caused be any non-uniform scale on StaticBodies. So when we scale a BeltConveyor for instance, for a brief moment the ConveyorEnds are stretched, then our scripts rescale them back to uniform scale. The error message is describing that moment that the scale is invalid. And for that brief moment the collision shape effectively has the dimensions described in the message. Since our script fixes the scale, we won't run into any surprises with physics, so we really could just keep suppressing the error. Though it would be better to fix it so users could get warnings about their own mistakes.

RigidBodies didn't appear to have this problem at all.

closed 2025-05-03 by ryevdokimov