Open-Industry-Project/Open-Industry-Project-Godot #81: Change the width on the individual chain in the chain transfer

ryevdokimov, 2024-10-16:
### Discussed in https://github.com/orgs/Open-Industry-Project/discussions/80

<div type='discussions-op-text'>

<sup>Originally posted by **tiagocunico** October 16, 2024</sup>
On the current release, I'm not able to simulate a real system with the options on the chain transfer configuration.

With this configuration:
[image]
I got this behavior: (the box tilts when been transfer)
[image]

To make it tilt less, I changed the code to be able to build this chain transfer:
[image]
And got this behavior: 
[image]
It works better, but the problem with this solution is that it is heavily unoptimized, and sometimes the box still tilts when transferring. I think we could solve this problem by been able to increasing the width of the individual chain, so the box doesn't have the space to fall between chains.</div>

cooperra, 2024-10-16:
This feels like a physics issue. Like, I wonder if it would work better if the box was heavier or moving slower. Looks like we currently have all boxes set to 10 kg (22 lbs).

Anyway, if the goal of this issue is to allow tighter chain spacing like above, then that's an easy fix. We would just lower the minimum `Distance` clamp. And if we add a range property hint, we can continue to have a slider with a "soft minimum", like the distance we have now, that the user can override by typing in a smaller value.

tiagocunico, 2024-10-16:
I think it would work better if the box was heavier, I'm used to working with 30kg box and I have never seen the box do this kinda of tilt on this type of transfer, maybe the difference is the center of mass. I didn't though I could change the mass of the box.

I think the problem with just lowering the minimum distance clamp, is that for longer boxes it will be very heavy on the performance of the pc, I had to use 2 transfers with 29 chains with 0,01 distance to work, and the simulation was very laggy.

ryevdokimov, 2024-10-16:
> I didn't though I could change the mass of the box.

You can, but since you mention it. It's probably worth exposing this to the part, so it can be modified on a per instance basis. Might include it as a property in `BoxSpawner` as well (and include randomizing it?)

>I had to use 2 transfers with 29 chains with 0,01 distance to work, and the simulation was very laggy.
@cooperra , possibly a scaling every frame issue like the rest?

cooperra, 2024-10-17:
Probably. It also doesn't help that chains are still RigidBody in master
and 1cm is close enough that they'd overlap. I doubt the physics engine
would like that very much, but it might be fine since they're frozen.

PR #122 (2024-12-18): Allow tighter ChainTransfer spacing
Distance between chains can now be reduced to 3cm (previously 25cm).

3cm is the width of a physical chain, so a 3cm Distance is the smallest possible without overlapping them.

The PropertyHint slider still uses 25cm as the minimum (the distance where the container models touch), but the user can type in a smaller number.

Closes #81.
closed 2024-12-18 by ryevdokimov