Open-Industry-Project/Open-Industry-Project-Godot #96: Implement a "walk mode" to allow users to explore a scene like a person

ryevdokimov, 2024-10-25:
Would be a nice alternative to flying around.

cooperra, 2024-10-25:
You might be interested to know about **Editor Settings > Editors > 3D > Freelook > Freelook Navigation Scheme > "Fully Axis-Locked (Minecraft)"**

The downside is the Q/E keybinds for moving up and down aren't very discoverable in my opinion.

ryevdokimov, 2024-10-25:
That partially is what I would want, but I was thinking something more along the lines of an actual character body that can interact with the scene in a more intuitive way.

ryevdokimov, 2025-01-18:
See: https://github.com/godotengine/godot-proposals/issues/11424

ryevdokimov, 2025-02-25:
Closed by: aeb224634d39c75711d35209b0964853fda5beb7

Default shortcut is: `Shift + R` 

[image]

closed 2025-02-25 by ryevdokimov