Open-Industry-Project/Open-Industry-Project-Godot #219: How to run it?

bjorn-lundin, 2025-07-24:
Hi,
This project looks very interesting.
But how do I start it?

I'm on a Mac with apple silicon,
downloaded dotnet8 and then Open_Industry_Project.zip.
But then - what is next?
as below - dotnet thinks its a library - not an exe.

`bnlair:Open_Industry_Project bnl$ dotnet --version`
8.0.412

`bnlair:Open_Industry_Project bnl$ uname -a`
Darwin bnlair.local 24.5.0 Darwin Kernel Version 24.5.0: Tue Apr 22 19:54:33 PDT 2025; root:xnu-11417.121.6~2/RELEASE_ARM64_T8122 arm64
bnlair:Open_Industry_Project bnl$ 

`bnlair:Open_Industry_Project bnl$ dotnet run Open\ Industry\ Project.exe --project ./default_project/OpenIndustryProject.csproj `
Unable to run your project.
Ensure you have a runnable project type and ensure 'dotnet run' supports this project.
A runnable project should target a runnable TFM (for instance, net5.0) and have OutputType 'Exe'.
The current OutputType is 'Library'.

ryevdokimov, 2025-07-24:
Hello, we're planning on releasing a new version in the near future which will not require .NET, but unfortunately there is a known issue with running OIP on macOS: https://github.com/Open-Industry-Project/Open-Industry-Project/issues/145.

Since none of our current developers are on Mac it is very tricky to debug.

bjorn-lundin, 2025-07-24:
Ok, then it is not me who is the problem :-)
Hmm, I'll try on a win-computer this weekend instead then.
Thanks for the feedback.